RoboticsBrno/RB3204-RBCX #2: [CZ] MainBoarsSCH - dodělat/zkontrolovat

JarekParal, 2020-03-08:
## Dodělat

### Sheet Power
- [ ] vybrat výkonové tranzistory a signálové tranzistory 
- [ ] doplnit atributy u tlačítek a ověřit dostupnost na JLCPCB

### Sheet Motors
- [ ] @cednik dodat LCSC_ID pro BAS40-05, SS34, @JarekParal doplní atributy

## Zkontrolovat
- [ ] OZ INA180A4 - ověřit správnou velikost pouzdra - ve schema je SC-70-5 a v [datasheetu/JLCPBC SOT-23-5](https://datasheet.lcsc.com/szlcsc/1811211019_Texas-Instruments-TI-INA180A4IDBVT_C133969.pdf) což [jsou zdá se různá pouzdra](https://en.wikipedia.org/wiki/Small-outline_transistor#SOT23-5.2C_SOT353)

## Otázky
- [ ] jak byla vytvořena značka `+3V3 STM`?
- [ ] proč byli odebrány tlačítka v listu `Coprocessor`

## Podměty

### Atributy
- dolňujme atributy u součástek, hned jak součástku vybereme a začnemé dávat do schema (ideálně vyplnit u první součástky a pak už jen rozkopírovat)
- doplňované atributy: Description, Manufacturer (MF), MF_ID, LCSC_ID, Package, Price (PRICE_1PLUS, PRICE_30PLUS, PRICE_100PLUS...), Extended_part Yes/No

cednik, 2020-03-08:
- INA180A4: máš pravdu, pouzdro opravím
- `+3V3 SMT`: knihovna supply - jsou v ní všechny použité symboly napájení. Celkem bude na desce 5 různých "globálních" napájení (nepočítám napájení pro serva, které je ve schématu u sebe, a podobné): beteriové napětí `VCC`, `+3V3S` a `+5V0S` napájení pro STM, které musí fungovat nezávisle na ESP32 (pokud toto na desce zrovna nebude osazené) a `+3V3` s `+5V` napájecí napětí pro ESP32 a věci k němu připojené. `GND` je pro všechno společné. Bez zvláštních symbolů by v tom byl hrozný zmatek.
- Odebrání tlačítek: ten symbol byl příšerný. Nakreslil jsem hezčí. Zatím je použit v listu napájení. Do zbytku doplním, jakmile se k daným listům dostanu.
- Atributy u odporů a kondenzátorů zatím nejsou záměrně. Aktuálně jsou u součástek napsané jejich optimální hodnoty z pohledu elektrického (aspoň dle mého názoru :-) ). Až bude schéma kompletní, je potřeba ho znovu projít a udělat optimalizaci kvůli ceně a množství použitých hodnot. Pak má smysl doplňovat atributy, nejlépe nějakým skriptem :-)

JarekParal, 2020-04-12:
Udělal jsem si menší rešerši na velikost DPS, počet vrstev a případné cesty jak ušetřit místo.

## Velikost DPS a počet vrstev versus cena

yaqwsx, 2020-04-12:
Když se dívám na schéma, podle mě bez 4 vrstev nemáme šanci anebo u toho minimálně přijdeme o vlasy. O co přeplatíme na prototypech, o to rychleji (a lépe) to budeme mít hotové. Cena při výrobě 50 ks je v kontextu tábora zanedbatelná (rozdíl cca 1 USD). Rozdíl při objednání dalších desek (10 ks) - 0.5 vs 3 USD, mi přijde také akceptovatelný. Toť můj názor a počet vrstev.

Co se týká rozměru - já vždy preferuji co nejmenší desky, ale to je asi má osobní úchylka.

Ad svorkovnice: a) se mi nelíbí, b) c) jsou za mě ok.

dzarda, 2020-04-12:
Zaroutovat to určitě půjde, jde ale o komfort (vlasy) a rychlost, což nás momentálně trápí nejvíc.  Když jsme dělali RBC, bránil jsem se stejně. Nakonec to šlo, ale byl to "Sokoban". Ta cena 4 vrstev je pořád za hubičku, i pro 12x10.

PS nemá myslím smysl tahat GND cesty, to bývá nejlepší vyřešit polygonem - jen je u toho potřeba uvažovat o reálných tocích proudu.

JarekParal, 2020-04-14:
@dzarda Na RBC byl rozdíl v tom, že u driverů jsi prakticky měl 4 vrstvy, protože byla na samostatné DPS. Zároveň tady je kolem driverů mnohem víc součástek kvůli měření proudu.
Ale možná už jsem vymyslel, jak by se to dalo zaroutovat i na 2 vrstvách, když se upraví dispozice motorových svorkovnic.

cednik, 2020-04-14:
Když koukám na Jarkem zaroutované napájení, pojďme do čtyř vrstev. Cenový rozdíl není tak velký - naše vlasy jsou dražší.
Honzovu úchylku na minimální rozměry desky sdílím.

### Svorkovnice

- Je to sice hezká feature, ale jejich úplnému zrušení bych se kupodivu nebránil. Pokud se povede udělat destičku s enkodérem, která se na motor jen připlácne a kontakty motoru se zapájí přímo do ní, pak budeme motory stejně připojovat jedním šestižilovým kabelem.
- Varienta **b** se mi nelíbí. Za prvé si nemyslím, že to moc pomůže, za druhé narazíme na proudové limity - jak u svorkovnice, tak u připojovaných drátů (ten nejtlustší drát, co jde strčit do @JarekParal návrhu č. 1 sice zvládne 10 A, ale ohřeje se přitom o 40 °C [[wiki:awg](https://en.wikipedia.org/wiki/American_wire_gauge#Tables_of_AWG_wire_sizes)]).
- S variantou **c** bych se smířil. Pár milimetrů místa ušetří, drivery půjdou posunout víc ke kraji desky a budou víc zarovnané se zapínáním uprostřed s velkými svorkovnicemi.

Ovšem nelíbí se mi šroubovací svorkovnice v rozteči 2.54 mm - potřebuješ na ně hodně malý šroubovák, což bývá často problém. Jen nevím, jestli ty navrhované pružinkové dvoudírové ušetří nějaké místo v tom správném směru (šířka/hloubka té svorkovnice). Místo, ušetřené její délkou, mně moc nenapadá, jak využít.

P.S.: @dzarda tahání GND cest = uvažování o reálných tocích proudu. Ten polygon tam pak stejně bude, ale budeš si jistý, že proud má dost místa k tečení a nepřehlédls někde propojení jen nějakým chlupem.

JarekParal, 2020-04-28:
Svorkovnice pro motory byli z návrhu odebrány kvůli nedostatku místa na DPS. 

Pinheady pro enkodéry (a motory) jsou nahrazeny [JST PH 2.0 mm konektory](https://www.aliexpress.com/item/32810190369.html), jelikož tyto konektory používají i [čínské motory s enkodéry](https://www.aliexpress.com/item/32944478481.html), dají se k nim koupit [snad kvalitní nakrymplované kabely](https://www.aliexpress.com/item/32890983007.html), které nejdou otočit.

RBCX bude navrhována jako 4 vrstvá DPS, protože ve dvou vrstvách se nelze vymotat. Ale cílem je zachovat rozměry 10x10 cm.

dzarda, 2020-05-03:
Přemýšlel jsem nad option jumperama - nechcem radši nahradit za 0R součástky s defaultním výběrem jedné varianty, ať odpadne manuální postprocessing?

yaqwsx, 2020-05-04:
Varianta nechat si tam u JLC osadit rezistor se mi hodně líbí. Zvládneme udělat footprint tak, že zůstane zachovaný současný tvar pájecích plošek a JLC nám do něj korektně umístí 0R odpor (@cednik @JarekParal)? Proč to chci:
- získáme defaultní konfiguraci z výroby (juch)
- když to chci změnit, přejedu to pájkou, setřu odpor, a stále mám věc, která se pohodlně pájí, protože je velká. Kdybych tam měl pady přímo pro 2 rezistory, tak ta změna bude fakt bolet.

JarekParal, 2020-05-04:
Včera jsme @dzarda návrh řešili a taky se nám líbí. Je tam jeden zádrhel.

Pokud chceme nulové výkonové odpory

cednik, 2020-05-04:
Defaultní konfigurace je super nápad. Včera jsme se o tom bavili s @JarekParal po Skypu.
Asi by neměl být problém nechat na stávající footprint osadit 0R 0603, akorát budu muset trochu překopat tu součástku, abych to do osazovacích dat nemusel dopisovat ručně.
Momentálně mám nastavovací jumpery ze spodní strany desky, aby byly přístupné ze stejné strany, jako konektory. Kvůli defaultnímu osazení bych je musel hodit nahoru a pak se k nim v robotovi nepůjde dostat bez odšroubování desky, ale to asi není tolik na škodu.

Jaké chceme defaultní hodnoty (návrhy):

- **SPI**: dvě jsou pevně 3V3, jedno nastavitelné (ale pouze napájení, na signálech level shifter není) @yaqwsx jaké napájení chce tvoje senzorická lišta?
- **I2C**: dvě 3V3, dvě nastavitelné (včetně level shiftrů) - navrhuji **5V**
- **UART**: **3V3**
- **UART2**: (JP40, stejný pinout jakou UART, co si tam uživatel v ESP naprogramuje je na něm - 2. UART/I2C/CAN/cokoli) **3V3**
- **UART koproc**: (JP33) **5V** - STM má 5V tolerantní piny
- **enkodéry**: **3V3** - podle vybraných hallových sond
- **konektor komunikačního UARTu mezi STM a ESP**: (JP54, dětem se stejně nebude osazovat) - **neosazovat**
- **SWO signál na SDW konektoru STM**: možnosti jsou debug UART TX / SWO pin. SWO pin je bohužel sdílený se signálem jednoho enkodéru. Jde prakticky jen o pohodlnost, debug UART je vyveden na samostatném konektoru.

Schéma konektorů:
[image]
[image]

yaqwsx, 2020-05-04:
Lišta by měla fungovat jak na 3V3, tak na 5V (je třeba jen správně osadit rezistory pro LEDky). Čip je 2.7-5V: http://ww1.microchip.com/downloads/en/DeviceDoc/21295d.pdf

JarekParal, 2020-05-19:
Done #15 and 14de8e774d4055421ac026269bed3bde5665a5fa

closed 2020-05-19 by JarekParal