RoboticsBrno/RB3204-RBCX #6: CI + Code quiality

JarekParal, 2020-04-09:
What do you think about adding CI for test builds and some Code quality tool, for checking the source code?

Also, we should use the [rb_clang_format](https://github.com/RoboticsBrno/rb_clang_format) and all convention from WebKit which we approved.

@dzarda @yaqwsx @cednik @Tasssadar

dzarda, 2020-04-09:
First step should be just building the fw. Automatically checking code style seems tricky at this point IMHO

JarekParal, 2020-04-09:
Yes, the first step is building ("adding CI for test builds"). But that should be easy with PlatformIO. We already have that on many repositories. 
Why do you think that checking code style is tricky? Also, we can use it just as advice tool. Not strong requirement.

dzarda, 2020-04-09:
Tricky because I don't have best experience with it from work. Stuff fails on CI but not locally. Could be better with our workflow. Also there will be code in the repo (already is) that's not ours. Analyzer should ignore it.

yaqwsx, 2020-05-04:
>Stuff fails on CI but not locally.

I think it is actually a good thing. CI verifies that you are not using some nuance of your local environment which the others cannot reproduce. Also CI "is the reference environment" which is easy to track and reproduce. Just to be clear - with PlatformIO there shouldn't be such problems.

To easily debug failures in remote, I started to run all my Travis CI in docker, which I can easily execute locally - see [an example](https://github.com/paradise-fi/RoFI/blob/master/.travis.yml). Docker also makes easy (and validated) tracking of development dependencies (see [corresponding Dockerfile](https://github.com/paradise-fi/RoFI/blob/master/Dockerfile)).

dzarda, 2020-05-14:
Can we close this?

JarekParal, 2020-05-15:
Yes, let's close this issue. CI is integrated and Code quality wouldn't be soo easy.

closed 2020-05-15 by JarekParal