RoboticsBrno/RB3204-RBCX #13: Coproc control communication

dzarda, 2020-05-02:
We decided to dedicate one UART between ESP32 -> STM32 for control communication:
- Asynchronous bidirectional control/state transfers
- Frames containing Protobuf messages

### Link/PHY
Probably some prefix+length as packet start and hope for the best. Do we need some timeout logic for framing?

Gonna start with 115200 since that's the limitation of my dev HW. What about using the *parity bit* as a simple error detection?

### Endpoints
Let's get inspiration from USB. Each request packet should contain some endpoint identification. All endpoints have their ID and type associated with them. This association (in case a status/response message) may be described in protobuf as follows:

```protobuf
message CoprocStatus {
  oneof payload {
    fixed32 protoVersion = 1;
    fixed32 fwVersion
    float batteryVoltage = 2;
...
    MotorStatus motorAStatus = 32;
    MotorStatus motorBStatus = 33;
...
}

message MotorStatus {
  float current = 1;
  sint32 rotation = 2;
  sint32 velocity = 3;
...
}
```

Ideas welcome

yaqwsx, 2020-05-04:
About link/phy: Since the packets are always received via UART/USB which have strict timing (unlike network), the timeout of several characters/several USB frames is IMHO welcomed. When the user sends an incomplete message, it does not cause the failure of the next one.

I am not sure if I follow the idea from the example. Could you explain it in a different way? Or let's have a call.

To present my view: there are two ways we can go:
- flat commands
- nested/namespaced commands

For the rest of the text, consider we have 2 motors, we want to turn them on/off and we have 2 buttons and we would like to query their status (presses/released).

Flat commands are commands like "turn motor on", "turn motor off", "get status of a button". And they are parametrized by component idx. They are all distinct and simple. Imagine you have a large enum of all possible command types and there is a large dispatcher in the code.

Nested commands are grouped - you have namespace motors (providing command turn on, turn off parametrized by motor idx) and namespace buttons (providing command query parametrized by button idx). Imagine you have 3 enums: namespace, motors commands, button commands.

TLDR: I prefer nested commands.

The obvious advantages for nested commands are:
- there is no huge dispatcher in you code and instead, there are multiple smaller ones that dispatch messages in between each other until one of them performs the command. Which is neat.
- documentation for such protocol is IMHO more readable - you know where to look when you want to control motors
- it is easy to come up with command names
- you can easily add new/remove functionality with local changes in the code.

There are some disadvantages:
- for simple tasks, it adds a little bit of complexity for the user.
- it is unclear what structure should the responses follow (should they also be namespaced?)
- inter-namespace commands do not fit this pattern well, unless (solution at the end of the paragraph). Consider you would like to have a command "When button x is pressed turn motor y" and command "Disable previous functionality". With flat commands this is easy - you introduce two new commands and implement two functions. With nested commands, there is a question - to which namespace it belongs? Motor or button? If we preserve dispatcher isolation, none of them can implement such functionality. Imho, the only solution is to introduce new namespace (with a new dispatcher) - e.g., namespace "chain", which provides command "tie" taking button idx and motor idx.

BTW: Now I am thinking, that dispatcher is a stupid name and maybe we should call it simply a controller.

dzarda, 2020-05-07:
We've got nice results here https://hackmd.io/bpK-FZ-MRfOS4JP8o0kNZg?edit

dzarda, 2020-09-04:
Closing, this is a done deal

closed 2020-09-04 by dzarda