RoboticsBrno/RB3204-RBCX #14: Coproc packet framing

dzarda, 2020-05-02:
Dedicate a new UART as a transceiver of packets with a fixed prefix.

Nanopb provides length-prefixed encode/decode, so this doesn't have to be solved here.

Required for #13

yaqwsx, 2020-05-04:
Is there a suitable counterpart of the encoding for desktops? Or do we have to implement it ourselves on the desktop?

Otherwise, I agree with usage of encoding provided by Nanopb.

dzarda, 2020-05-04:
We'll need to make a converter for desktops. I'm thinking like

There's things that implement the COBS encoding like libspf, but I think that's an overkill. PB3 doesn't send zero values, and zero bytes shouldn't appear on the wire unless present in 'bytes' and the like

So I think we may simply use 0x00 as the packet terminator. Together with some timeout, it should be robust enough IMO.

dzarda, 2020-05-04:
> and zero bytes shouldn't appear on the wire

Of course I messed up somewhat - the `fixed` types will contain zeros verbatim

dzarda, 2020-05-14:
More thinking done - I feel like I want the COBS encoding so that zero bytes are only present as packet terminators. Would allow time invariance in receive logic and better robustness imo.

Seems like there's things available OOTB for C and py.

yaqwsx, 2020-05-14:
You are more experienced in this area and by briefly looking, it looks nice. So, go for it and let's use it.

PR #17 (2020-05-18): Add control framing
- Allocate "secondary" USART for control frame comm
- Add control frame loopback
- Use [cobs-c](https://github.com/cmcqueen/cobs-c) for COBS (to avoid zeros on the wire except frame start)

Closes #14

#### Frame format

```
0x00 0xLEN COBS-ENCODED-DATA
```

Max overall frame length is 257, COBS data limited to 255 bytes. From the nature of COBS, the first byte is wasted, therefore maximum payload is limited to **254 bytes**. There are always 3 bytes of overhead.
closed 2020-05-18 by dzarda