RoboticsBrno/RB3204-RBCX #24: Zprovoznit programování přes tunel

dzarda, 2020-06-07:
USB tunel už je v STM implementován pomocí dmitrystu/libusb_stm32. Má to ale lehké mouchy:
- V downstream kanálu dochází ke zdrátě dat, když se naráz pošle víc jak 64b (velikost EP). Řešeno v https://github.com/dmitrystu/libusb_stm32/issues/51
- Je nastaveno fixně 115200, lze řešit v `cdc_control ... case USB_CDC_SET_LINE_CODING:`
- UART je stále aktivní nezávisle na připojení CDC

Asi nejdřív třeba vyzkoušet, jestli to v aktuálním stavu už náhodou nejde naprogramovat.

Tasssadar, 2020-06-08:
Ano, a chtěl bych tam přidat i to nastavování rychlosti UARTu, jinak to bude těch 115200 při programování brzdit.

cednik, 2020-06-08:
To by bylo super 👍
Jinak při ovládání těch pinů je potřeba implementovat následující tabulku [[1](https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch.pdf)]:
[image]

Pak je potřeba si pohlídat úroveň na pinech [[2](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf#4)]
- STM PB15(SPI_MOSI/iLed) - ESP IO12 (MTDI/iLed): LOW
- STM v1.0 PC11(UART4_RX/control) v1.1 PD9(UART4_RX/control) - ESP IO2: LOW

PR #31 (2020-06-15): Fix CdcUartTunnel, implement ESP32 boot sequences
S tímhle funguje programování přes tunnel, včetně nastavování rychlosti & spol.

* Tunnel testován pomoci `fw/tools/looptest.py` na 115200 a 921600 baudu, tunnel neztrácel žádná data po nejméně 10MB.
* Co se týče náhodných znaků vrácených při otevření tunnel na Linuxu zpět do STM, částečně vyřešeno tím, že zahazujeme data, pokud si je PC nevyzvedne dostatečně rychle (tj. když port není na PC otevřený). Přijde mi, že to "FTDI" co ja na ESP32 devkitu to dělá stejně.
* Programování (pouze) testováno na dvou PC, 1x na Windows a 2x na Linuxu

Closes [... cut]
closed 2020-06-15 by dzarda