RoboticsBrno/RB3204-RBCX #28: hw_v1.1 changes

cednik, 2020-06-08:
- [ ] Posunout dráhy měření proudu u serv dál od elektrolytů
- [ ] Upravit pinout
- [ ] Opravit popisky po změně pinoutu
- [ ] Zmenšit zapínací kondenzátory
- [ ] Opravit fungování zapínacího tlačítka bez připojené baterky (napájení z USB)
- [ ] Přidat Bmid na svorkovnici
- [ ] Přidat rezistor mezi STM a power
- [ ] Posunout součástky dál od pinheadů, kvůli pájení
- [ ] Opravit masku u DRV8833 center padu
- [ ] Zvětšit testpoit GND tak, aby do něj šel zapájet pinhead
- [ ] Přesunout popisky pod piezzem
- [ ] Nahradit GND_SEPARATOR cutout polygonem
- [ ] Zmenšit velikost nastavovacích propojek
- [ ] Upravit polygony napájení serv
- [ ] Posunout testpointy dál od kondenzátorů
- [ ] Zkontrolovat thermaly
- [ ] Upravit rozlití mědi kolem upevňovacích otvorů
- [ ] Lehce roztáhnout piny USB-C
- [ ] (Prohodit micro USB pinheadem, aby bylo místo na kabel vedle USB-C)
- [ ] (Vytáhnout buzzer na pinhead)
- [ ] (Vytágnout TP62 zpod ESP)
- [ ] (Přidat tranzistor na zapínání iLed step-downu)
- [ ] (Přehodit I2C na konektoru pro expandér)

yaqwsx, 2020-06-09:
[ ] Zvážit odebrání grafiky na vstvách tPlace a bPlace ve footprintech. Podle [tohoto odkazu](https://electronics.stackexchange.com/questions/186324/what-is-the-difference-between-tplace-and-tdocu-layers-in-eagle-pcb) jsou tPlace a bPlace míněny jako vrstvy silkscreenu a KiCAD import je tak uvažuje. Jedná se o obdélníky okolo koprocesoru:

[image]

yaqwsx, 2020-06-09:
- [ ] Zvážit nepoužít mezery pro zarovnání textu, ale mít každý sloupeček jako samostatý text; rozbíjí KICAD import kvůli lehce jinému fontu:

[image]
[image]

cednik, 2020-06-09:
- [ ] Popisky svorkovnice zezadu (zepředu pod ní nejsou vidět)
- [ ] Přidat přepěťovou ochranu na měření prostředního článku

cednik, 2020-06-15:
- [ ] Připojit gate Q8 na ESP_EN - fix pull-upu na IO12.

dzarda, 2020-06-17:
[ ] Zvážit přidání pájecích ploch zboku USB konektorů pro lepší odolnost

[image]

cednik, 2020-07-08:
Verze v1.1 vyrobena, zavírám.

closed 2020-07-08 by cednik