RoboticsBrno/RB3204-RBCX #33: Deploy FW na roboty

Tasssadar, 2020-06-16:
Jakým způsobem chceme dostávat FW na desky pro děti? 

1. Při oživování, normálně STLinkem
2. ???

cednik, 2020-06-16:
Při oživování normálně STlinkem.
Další možnost je embeded bootloader skrze USART1 (USER).
V souvislosti s tím jsem přemýšlel, jestli neprohodit USER a CONTROL, ale asi se mi to nelíbí.
Krásný by byl USB bootloader 😇

dzarda, 2020-07-01:
Tohle by mohlo být fajne https://github.com/dmitrystu/sboot_stm32

dzarda, 2020-07-07:
Mega ultra by byl autoflash přes ESP při startu aplikace

Tasssadar, 2020-07-07:
Bylo, ale bohužel to neřeší první flash, ten stejně asi bude muset být vždy ST-linkem :/

dzarda, 2020-07-08:
Asi bude vhodnější to dělat přímo z PC (přes DFU nebo vystavíme control rozhraní i do PC)

closed 2020-07-16 by Tasssadar