RoboticsBrno/RB3204-RBCX #52: Defaultní 5V zdroj

dzarda, 2020-07-20:
Deska by imho měla mít výchozí zdroj pěti voltů. Práce s levnými stepdowny zdá se mi nešťastná pro většinu případů.

yaqwsx, 2020-07-21:
V rámci https://github.com/RoboticsBrno/RB0004-NeopixelBooster jsme s Martinem Kousalem navrhli 3A step-down čistě z součástek z JLCPCB, který se cenově vleze do 1 USD a plochou taktéž není náročný.

cednik, 2020-07-21:
Bylo by to krásné, ale i když malý, jsou to 2 cm², které tam teď nejsou.
Znamená to desku celou překopat.
Jsem pro to udělat a zároveň implementovat #51.
Jen se musíme rozhodnout, jestli vznikne RBCX 2.0, nebo RBCX+ :-D

dzarda, 2020-07-21:
Za mě stačí bump na verzi 1.2 :P

A nestačilo by čistě náhodou prostě LDO?

cednik, 2020-07-21:
Přesně tak. LDO by nejspíš stačilo na napájení STM, ale na ostatní periférie (včetně ESP) už chceš spínaný zdroj.
Pouhá výměna stabilizátoru a ostatní moje issue k verzi 1.1 ( #45 )  stačí na v1.2. Přidání nabíjení už je podle mě moc velký zásah.

dzarda, 2021-05-19:
Skvělý by byl i OC výstup, kterým by šlo uživatele informovat o možnosti osadit další stepdown modul :P

cednik, 2021-05-19:
Co myslíš OC výstupem a při jaké příležitosti bys chtěl takovou informaci podávat?

dzarda, 2021-05-19:
Myslím nadproud na stepdownu. Podávat by bylo vhodné až uživatelovi "operátorovi", společně s např. napětím baterky