RoboticsBrno/RB3204-RBCX #54: Real time clock

dzarda, 2021-03-31:
Jirka potřebuje rozchodit RTC na naší úžasné desce.

- Přidat PB zprávy
- Přidat RTC driver
- Otestovat RTC, změřit odběr...

dzarda, 2021-04-05:
RTC nemá kalendářovou logiku, ale umí pouze monotonicky čítat. @Tasssadar Stačí umět čítat sekundy?

- Set zpráva nastaví sekundy (implikuje inicializaci)
- Get zpráva přečte sekundy (nebo 0, když nemáme baterku/není inicializování.

yaqwsx, 2021-04-05:
Nestačilo by nám mít unix epochu?

dzarda, 2021-04-05:
Jo, však to je ono. Klient určí hodnotu v Set funkci, od toho to pak počítá sekundy. Naskýtá se mít to jako unix timestamp

Tasssadar, 2021-04-05:
To zní jakože by mělo stačit, "set" může nastavit buďto epoch, nebo prostě cokoliv. Vyšší přesnost než vteřiny neumí?

dzarda, 2021-04-05:
Umí to přesnost jakou chceš, ale má to jen 32 bitů

dzarda, 2021-04-06:
Změřeno 1,26uA

cednik, 2021-04-06:
> 
> 
> Umí to přesnost jakou chceš, ale má to jen 32 bitů

Běží to z krystalového oscilátoru 32.768 kHz, tudíž 1 sekunda je myslím nejvíc smysluplný časový úsek, který má smysl z toho dolovat. Ale předděličku si můžu nastavit od 1 do 1 048 576 (2^20), takže kdyby někdo moc chtěl, může to být přesnější.
Ovšem pak se špatně počítá ta epocha, když to může přetéct, zatím co procesor nemá šťávu :-)

cednik, 2021-04-06:
Kromě get a set bych ještě poprosil o alarm.get/set/enable/disable.
Ono to totiž pak umí přes POWER pin (PC13) celé RBCX v nastavený čas zapnout :-)

dzarda, 2021-04-06:
Najs, alarm udělám

dzarda, 2021-04-13:
Alarm bude vhodné nastavovat asi jako parametr `powerOff` příkazu, protože nastavení `ASOE` tu nohu okamžitě stahuje, čímž vypíná desku.

Protože to stahuje dolů furt i po vypnutí, zapnutí tlačítkem nefunguje - jen problikne a zase se vypne. Musím zkusit, jestli během probliknutí nastartuje proc a jestli je čas tu nohu odpoutat.

cednik, 2021-04-14:
Mám za to, že jsem to zkoušel a že to fungovalo. Bohužel, v testovací větvy v commitech jsem to nenašel :-(

Nastavovat alarm jako parametr powerOff klidně, ale asi se mi to moc nelíbí jako jediná možnost - asi si chci umět nastavit alarm dopředu, dřív, než budu zařízení vypínat. Nehledě na možnosti že se to vypne samo kvůli vybité baterce, nebo jejímu odpojení.
Nemůže si STM nastavený alarm pamatovat a až při vypínání nastavit správně power pin? (Ať už se vypne příkazem, nebo poklesem napětí na baterce, ke kterému dojde i při jejím odpojení.)

dzarda, 2021-04-14:
Jasně, musí si to vlastně pamatovat stejně kvůli těm průběžným vypnutím/zapnutím...

> zkoušel a že to fungovalo

Myslíš tu insta alarm deaktivaci při startu?

cednik, 2021-04-14:
Zkoušel jsem, jestli to tím jde zapnout a šlo to. Jestli jsem dělal deaktivaci si bohužel nepamatuji. Nicméně si pamatuji, že to umělo dělat jen pulz, nikoli držet úroveň, tak jsem ji asi dělat musel.