RoboticsBrno/RB3204-RBCX #68: Lepší motor driver

dzarda, 2022-08-15:
Stálo by za zvážení, zda do další revize nedát proudově lepší drivery motorů

yaqwsx, 2022-08-15:
Na Serviu používáme **DRV8870**
 - jednokanálový
 - 3.6A špičkově
 - 6.5-45V, má to interní omezení proudu.
 
 Já sním o **DRV8874**:
- které je jednokanálové,
- 6A špičkově,
- 4.5-37V
- umí měřit proud na úbytku na interním MOSFETU a zdrcadlit ho (tj. nepotřebuje current shunt, který topí)
- bohužel, dlouhodobě nedostupné. Mouser naskladní až 8/3/2023
- zatím se dá koupit za 2k devkit: https://cz.mouser.com/ProductDetail/Texas-Instruments/DRV8874EVM?qs=mAH9sUMRCtsW213NeE2uzA%3D%3D

dzarda, 2022-08-23:
U DRV8833 (co tam teď je) se mluví o paralelním zapojení výstupů. Pokud by se nám povedlo elegantně umožnit spojení výstupů na desce (nebo kabelem), mohla by to být cesta k řízení dvou lepších motorů.

[image]

[image]

JarekParal, 2022-08-23:
Mám za to, že jsme s tím při návrhu RBCX (možná už při RBC) počítali jako s možností jak uřídit silnější motory. Ale myslím, že žádná speciální příprava na DPS udělaná není.

cednik, 2022-08-23:
Taky si myslím, že se s tím jako s možností počítalo.
Na desce opravdu žádná příprava není a ani nevím, kam by se případně vlezla.
Softwarově je potřeba zaručit, že budou opravdu generována stejná PWM pro obě půlky (nevím, jestli teď FW v STM používá u časovaču dvojité bufferování?). Hardwarově doporučuji dva samostatné kabely, spojené až u motoru - v případě omylem nastavených různých PWM budou sloužit jako odpor a indukčnost a aspoň lehce omezí zkratový proud.

dzarda, 2022-08-23:
Double buf nevím, ale zřejmě je řešitelné. Double kabely zní asi dobře.