RoboticsBrno/RB3204-RBCX #71: Napájecí/stepdown modul

vachajirka, 2022-11-03:
Vznikna nabíjecí/stepdown modul, který bude zapojen do napájecího systému desky. Budeme chtít nabíjet baterku z USB. Modul bude
- USB budeme chtít přes QC na vyšší napětí
- VBUS_DET zaclampovat
- Střed baterky přidat na pinhead
- Vytěžit signál pro charge-enable
- Je třeba prozkoumat možnost implementace QC na hlavním koprocesoru (potřeba signál/y pro QC)

Když je ESP vypnuté, tak chceme být schopní nabíjet.

yaqwsx, 2022-11-03:
[image]

[image]

@dzarda: Tady jsou dvě mé implementace QC. Ta první funguje 100% (je vlastně i na původní nabíječce k RBCX), ta druhá by měla také (je to nápad od @cednik) ale netestováno (ale mám k dipozici desku, do které stačí jen QC logiky naprogramovat).

Zajímavé odkazy:
- https://github.com/Crypter/QC3Client/
- http://blog.deconinck.info/post/2017/08/09/Turning-a-Quick-Charge-3.0-charger-into-a-variable-voltage-power-supply

dzarda, 2022-11-06:
@cednik bude snad přímočařejší najít 3 signály (snad 2 QC + 1 charge-enable), než dávat nabíjecí procesor.

yaqwsx, 2022-11-06:
A což takhle přecijen přidat nějaký mini procesor, který by poslouchal na 1-wire? Nebo si prachsprostě na QC použít posuvný registr.

PS: To, co jsem nezmínil - určitě je třeba pro implementaci QC ještě měřit napětí na VBUS. Protože QC3 funguje po inkrementech.

dzarda, 2022-11-06:
Pokud nenajdem piny, tak expander/pos. registr zní dobře. Mohlo by být na I2C.

Měření vbus je pravda, asi logicky může nahradit VBUS_DET

dzarda, 2022-11-08:
Když je vypnuto a zapojím nabíječku, mělo by začít nabíjet.

yaqwsx, 2022-11-08:
Zajímavé téma ohledně čínských USB-PD/QC triggerů: https://www.eevblog.com/forum/projects/usb-pdqc-trigger-ic-anybody-know-what-these-aexp-modules-use/

yaqwsx, 2022-11-08:
...a možná by se dal udělat reverse engineering tohoto: https://www.aliexpress.com/item/1005004004234197.html?algo_exp_id=012a7090-c208-4754-95eb-061c2ca17f9d-30&pdp_ext_f=%7B%22sku_id%22%3A%2212000028680984029%22%7D&pdp_npi=2%40dis%21USD%211.57%211.52%21%21%210.57%21%21%402100bddf16679110452915034ec9e2%2112000028680984029%21sea&curPageLogUid=B7G8aBxbSLsv

yaqwsx, 2022-11-08:
A další várka IC (dostupné z LCSC a tedy i JLC PCB):
- 50 centů v kusovce: https://datasheet.lcsc.com/lcsc/2111121930_Legendary-LDR6328_C2919630.pdf
- pouze PD: https://datasheet.lcsc.com/lcsc/2006111335_INJOINIC-IP2721_C603176.pdf

dzarda, 2022-11-22:
Kandidát na 5V stepdown https://www.lcsc.com/product-detail/DC-DC-Converters_Texas-Instruments-TPS54623RHLR_C560970.html

dzarda, 2022-12-11:
Kandidát na nabíječku https://www.digikey.com/en/products/detail/monolithic-power-systems-inc/MP2651GVT-0000-Z/16688711

dzarda, 2022-12-13:
@yaqwsx Mohl bys mi prosím pomoct ještě s výběrem konkrétního QC/PD řešení?

Myslím, že USB BC rozpoznání nepotřebujeme. Asi stačí mít na začátku pokus o komunikaci...

```
pokusit se dělat PD/C current
pokusit se enumerovat USB
když USB enumerace vyšla:
   sosat 500mA (nebo víc pokud PD/C current vyšlo)
jinak když nevyšlo PD:
   pokusit se dělat QC
   když nevyšlo QC:
      sosat 500mA
```

Zajímavě vypadá CH224. Musel by se asi  od USB odstřihnout/začít napájet až po neúspěšné enumeraci,

LDR6321 taky umí asi velký věci. Dokonce tam vidím nějaký "EN" piny, ale nenašel jsem ani zmínku, jestli to má skutečně nějakou enable logiku.

yaqwsx, 2022-12-13:
Navrhuji:
- sehnat vzorky od obojího
- postavit s nějakým discovery kitem prototyp. 
Datasheety totiž vypadají docela mlhavě a nechce se mi riskovat, že nakreslíme desku, která nebude fungovat. Přemýšlím, jestli by náš problém ještě neřešil nějaký IC pro USB mux, který by nám DP/DM přepínal mezi procesorem a IC (anebo prostě QC handlovat v našem procesoru).

S procesem, který jsi napsal výše souhlasím a dělal bych to stejně.