RoboticsBrno/RB3204-RBCX #72: RBCX 1.3 featury

vachajirka, 2022-11-10:
- Využili bychom 6-ti vrstvou desku, abychom minimalizovali zásah do návrhu
- QC a možná i PD (ideálně integrovaný PD Sink IC typu STUSB4500 nebo PD PHY typu FUSB302)
- Nabíjecí/stepdown řešení viz. #71 , všechny stepdowny nahradit
- USB-C s lepším mechanickým jištěním
- Micro asi pryč

dzarda, 2022-11-19:
Import do Kicadu 6.0.9 funguje na první pohled překvapivě dobře. Zde regiony, které asi zasaženy.
[image]

yaqwsx, 2022-11-20:
Když napíšeš přesně co nefunguje, tak bychom to mohli doprogramovat.

dzarda, 2022-11-20:
Špatně jsem se vyjádřil, zatím jsem na žádné problémy nenarazil. Ten region je prostě to, kde očekávám změny.

dzarda, 2022-11-21:
Tohle vypadá hodně dobře pro PD+QC https://www.electronicsweekly.com/news/products/power-supplies/usb-type-c-sink-controllers-for-pd3-0-pps-2022-10/

yaqwsx, 2022-11-21:
Dívám se do datasheetu a nevidím tam zmínku o QC.

dzarda, 2022-11-21:
Doufal jsem, že půjde využít GPIO piny jako expandér :')